AgOpenGPS-Official/AgOpenGPS #614: FEAT: Resettable area counter(s)

gunicsba, 2025-01-26:
It would be great to have a running counter that we can reset. Similar to Trip A and Trip B in your car.

This would allow one when seeding or applying fertilizer to estimate how much is there until another fill up is needed.
The same could be used in the harvester to do a rough estimate of the rough yields. 

I believe something like this could fit in the advanced view of the Field Info button. All we need is an ability to reset at will.

Pat-I, 2025-02-03:
[image]

This one is not enough? It reset when you press it.

closed 2025-02-05 by richardklasens

gunicsba, 2025-02-23:
We tried this a couple times by now and we like every aspect of it.

Thank you for making it!