AgOpenGPS-Official/AgOpenGPS #616: Incorrect Altitude when NTRIP client in Use Manual Fix mode

ZombieFodder, 2025-01-29:
While doing some (unsanctioned hardware, i.e. derivatives of existing stuff) development, I noticed I was unable to connect to the [Ohio CORS VRS](https://www.transportation.ohio.gov/working/engineering/cadd-mapping/survey/cors-rtn) network for NTRIP corrections. The GPS part of my hardware was not working correctly (as I found out later). I reached out on Discourse (https://discourse.agopengps.com/t/ohio-cors-connection/18421) to figure out what I was going wrong. I was able to connect to my caster with other tools but not AgIO, so I (incorrectly) assumed there was an issue with the client built in. My error was based in the assumption that putting the NTRIP client in "Use Manual Fix" mode would ignore any data coming from the GPS receiver. I was searching all of the logs of traffic between my PC and Ohio's system trying to figure out what was causing the issues. I eventually figured out the issue was in the GPGGA message. 

### TL;DR
When in "Use Manual Fix" mode on NTRIP client, the coms between the PC and NTRIP caster do not populate the altitude when it comes in as zero from the antenna or never comes at all (HTTP to NTRIP Caster contains:)
`$GPGGA,230009.00,1234.211,N,01234.604,W,1,5,1,,M,46.4,M,0,0*..`

For some casters that use Virtual Reference Stations, this altitude is critical to the computation of corrections. As a result of Altitude not being populated, they do not transmit correction data. 

With the assistance of [FantasyTeddy](https://discourse.agopengps.com/u/FantasyTeddy), I believe I narrowed down the line that is causing my grief. 

https://github.com/AgOpenGPS-Official/AgOpenGPS/blob/0027bddb3638ba897c78ad4fe91e34bf95c98808/SourceCode/AgIO/Source/Forms/NTRIPComm.Designer.cs#L798

It might help (me, if no one else), if this was changed to something like:
```C#
//(may include syntax errors, I am not familiar with C#)
if (Properties.Settings.Default.setNTRIP_isGGAManual)
{
    sbGGA.Append("0.000,");
}
else
{
    sbGGA.Append(altitudeData.ToString("#.###", CultureInfo.InvariantCulture)).Append(',');
}
```
(may include syntax errors, I am not familiar with C#)

I appreciate everyone that put in so much effort to make this tool as cool/useful as it is, so I do not in any way expect this to be urgently addressed. I am happy to furnish any details relevant to the timely closure of this issue. Don't hesitate to ask for clarification if I missed important details.

gunicsba, 2025-02-02:
So in your example altitude data was missing and it ended up being ,, instead of ,0.000, ?

P00qwerty, 2025-02-06:
sbGGA.Append(altitudeData.ToString("#.###", CultureInfo.InvariantCulture)).Append(',');

should be

    sbGGA.Append(altitudeData.ToString("0.###", CultureInfo.InvariantCulture)).Append(',');

can you check if you need the decimals?

closed 2025-02-08 by P00qwerty

ZombieFodder, 2025-02-08:
> So in your example altitude data was missing and it ended up being ,, instead of ,0.000, ?

Sorry for the delay in my reply. Yes, the altitude was missing. Two commas with no characters in between.

commit 49f40bbf (2025-02-08): Fixes #616