AgOpenGPS-Official/AgOpenGPS #638: Danger of AutoSnapToPivot

chrisriteco, 2025-02-07:
[image]

Yesterday while working accidentally clicked this button (had to move bed a bit) and clicked the button in the menu, which activated this function immediately. After a few beds I found out that this was on, quite annoying. Maybe good to make it visually visible in the main screen that this function is on or that you can remove the button from the menu.

gunicsba, 2025-02-09:
For me in the tractor it's really hard to easily identify if its on or off. And as you said when it's on and you're not in some sort of plantation where you'd like to auto-snap it'll cause headache.

Maybe when enabled add some visual to the manual snap icon that it's also happening with the steer switch?

mattmccormack, 2025-04-12:
Perhaps the Auto Steer button icon could change to have the two arrows like the Auto Snap icon does?

[image]

[image]

closed 2025-07-29 by richardklasens