AgOpenGPS-Official/AgOpenGPS #664: Troublesome disengage configuration as Send and Save closes the form

gunicsba, 2025-02-23:
Here's why it hurts:
When we try to fine tune disengage we often play with the values. Set it low, then send+save, oh it's too low, lets increase it a bit.
Now we always have to reopen this form and navigate back.

richardklasens, 2025-03-04:
When changing these values like the pressure sensor you shouldn’t need to press send + save. The value will be updated immediately, and when ok, you press send + save. (The send is only needed for parts like the invert steer motor, dandoss etc.

richardklasens, 2025-03-04:
Correction, you're right, it will only update when press send + save, so i'm currently investigate to see if this can be send as soon as the value is changed (by leaving the form open)

gunicsba, 2025-03-13:
What's the reason to close the form?

Xilofeon, 2025-04-06:
It seemed like a good idea, but it was actually quite a pain to fine-tune the settings. I think it was fine as it was. (This is just my opinion)

richardklasens, 2025-07-30:
Fixed in #809

closed 2025-07-30 by richardklasens