AgOpenGPS-Official/AgOpenGPS #665: Migrating from older versions to 6.6+

gunicsba, 2025-02-23:
I noticed that on 6.7 the acquire / hold ratio is not calculated from the old settings files.

For example we used to have this:
```
           <setting name="setVehicle_goalPointLookAhead" serializeAs="String">
                <value>3.5</value>
            </setting>

            <setting name="setVehicle_goalPointLookAheadHold" serializeAs="String">
                <value>2.5</value>
            </setting>
```
I'd expect this value to be auto-calculated:
```
            <setting name="setVehicle_goalPointAcquireFactor" serializeAs="String">
                <value>1.11</value>
            </setting>
```
setVehicle_goalPointAcquireFactor = setVehicle_goalPointLookAhead / setVehicle_goalPointLookAheadHold;

FantasyTeddy, 2025-02-25:
This is a fair point, however I don't know of a good way how to do such a migration.

@AgOpenGPS-Official/aog-coders:
I think it is valid to say that migrating settings is something that will come up from time to time. Since we don't have the underlying mechanism from the `.settings` files anymore (which would have offered an [`Upgrade`](https://learn.microsoft.com/en-us/dotnet/api/system.configuration.applicationsettingsbase.upgrade) method), how could we achieve this?
What would be the requirements of such a migration?
- Does it need to be able to convert from any previous version to the current?
- Does it need to be fully automated/integrated?
- ...?

gunicsba, 2025-02-25:
At my previous company we used to have MigrationTasks that got executed on every project file. It basically analyzed / replaced values.
These had a date so they could be executed in sequence.

One such Migration Task here could be creating this value in case it's not present when opening a profile.

Not sure how feasible it would be here.

closed 2025-05-04 by richardklasens