AgOpenGPS-Official/AgOpenGPS #669: Inaccurate Field Distance calculations

J005t67, 2025-02-25:
When I want to 'Start New Field' via 'From Existing' or, 'Open' or 'Drive In', the corresponding window opens and show amongst others a table with one field per row. The values in the Distance column are however are inaccurate. I want to solve this before refactor this code to ViewModels

P-R9, 2025-02-28:
Hello, 
I am only user who saw that in the German Group there we talked about it. How you thought about it to make it better?
Would it be possible if when you drive on the field and it would recommand this field or it this to compex to add with the AOG code?
Like a popup window if you pass boundary.
[Do you want to open (Name of Field)]

As I got it right at the moment it takes one point as reference so maybe a good solution too, would be if you could set this point manuel as a user with something like a flag, because often you drive in from one side. And I guess this would be better to implement but would still be a big improvement.

richardklasens, 2025-02-28:
> Hello, I am only user who saw that in the German Group there we talked about it. How you thought about it to make it better? Would it be possible if when you drive on the field and it would [...]

Hi. 

If you’re standing in the field, and press field menu and hit drive in, it should open the field you’re standing in. (Except if there are more fields on the same spot, like using different fields on the same place). 
Setting the reference point manually won’t work. It’s needed to map the layer of your current area to the local plane used in AOG. The reference point will be stored where you created the field, or if imported from ISOXML/KML, centered in the middle of the current field.

P-R9, 2025-02-28:
Thanks for your answer,
It was only an idea thats the Problem often there are like many fields at one spot and than drive in does not work perfect as you said. So was just an idea and for me now it works good I think in older versions it didn’t work good if fields were from KML files.
If it could be set to the polygon or Boundary would be even better so it would could by itself without the list. But again I don’t know how complex it would be. I guess it hard to do. Now I only wrote  what some users in German group thought.
In small structures you get still many Fields shown and so it isn’t much faster than filter by distance and search in the List.
I know f.e fendt does it this way.
Was just what was in discuss so don’t need to implement it, as I know you put many work in it and it is still open source😁

xammmue, 2025-03-09:
Maybe another little input from my side.
Í often struggle a bit when using the drive in dialogue as it show a lot of fields for me. I understand, that this is most likely caused by my environment, as we have many small fields over a rather small area. For example 10 fields in a 500m x 500m square. Our fields are also only rarely largen than 1ha.

Scrolling through that list is not possible as far as I saw.

My idea would be to somehow make the distance treshold configurable. Maybe it also already is and I just didn't see it yet, I'll have to study the code some more in the future here. Having the list order by closest to farthest would probably also already help a lot

closed 2026-10-30 by richardklasens