AgOpenGPS-Official/AgOpenGPS #677: Introduce system/environment settings

FantasyTeddy, 2025-03-02:
In the current implementation, almost all configuration settings of AOG are being stored in the vehicle profile settings.
There are only a few exceptions which are currently stored in the Windows registry:
- Current working directory (must be known in order to retrieve vehicle profiles)
- Current vehicle profile (default profile to be loaded when starting AOG)
- Language (shared between AOG and AgIO)

Some of the configuration settings in the vehicle profiles do not really belong to a vehicle, but are more like system or environment settings, such as window sizes or display settings.

## Proposal

All of the more global system/environment settings should be stored in a global settings file (e.g. `%USERPROFILE%\Documents\AgOpenGPS\Settings.xml`)

### Which settings are system/environment settings?

This is the hard part, to figure out which settings don't actually belong into the vehicle profile settings.
Here is a proposal of all the settings that should be moved in my opinion:
- `setWindow_Location`, `setWindow_Size`, `setWindow_Maximized` & `setWindow_Minimized` (those are mostly application specific)
- `setMenu_isMetric` (is much like the global language setting)
- `setMenu_isGridOn`, `setDisplay_isStartFullScreen`, `setDisplay_isKeyboardOn`, `setDisplay_isAutoStartAgIO`, `setDisplay_isAutoOffAgIO`, `setDisplay_isSvennArrowOn`, `setDisplay_buttonOrder`, `setDisplay_isLogElevation`, `setWindow_isKioskMode`, `setWindow_isShutdownComputer`, `setDisplay_isShutdownWhenNoPower`
- `setDisplay_brightness`, `setDisplay_brightnessSystem` & `setDisplay_isBrightnessOn`
- `setMenu_isSimulatorOn`, `setGPS_SimLatitude` & `setGPS_SimLongitude`
- `setDisplay_colorDayFrame`, `setDisplay_colorNightFrame`, `setDisplay_colorSectionsDay`, `setDisplay_colorFieldDay`, `setDisplay_colorFieldNight`, `setDisplay_customColors`, `setDisplay_colorTextNight`, `setDisplay_colorTextDay` & `setDisplay_isDayMode`
- `setKey_hotkeys`
- `setJobMenu_location`, `setJobMenu_size`, `setWindow_steerSettingsLocation`, `setWindow_buildTracksLocation`, `setWindow_formNudgeLocation`, `setWindow_formNudgeSize`, `setWindow_abDrawSize`, `setWindow_HeadlineSize`, `setWindow_HeadAcheSize`, `setWindow_MapBndSize`, `setWindow_BingMapSize`, `setWindow_BingZoom`, `setWindow_QuickABLocation`, `setWindow_gridSize`, `setWindow_gridLocation` & `setWindow_tramLineSize`

## Questions

There are a few other questions that should be answered:
- Where to store these new settings? (suggestion as above: working directory + `Settings.xml`)
- How do we migrate to these new settings? (if we migrate at all)
  - The difficulty here is that we go from n to 1 (n vehicles to 1 setting) -> which vehicle defines the global settings?
- Where to move all these new settings in the GUI? (suggestion: add a new "Settings..." menu below "Language" and move all the settings there)

gunicsba, 2025-03-03:
For the migration:

Vehicle settings conflicts with Aog Settings.xml !!
Keep existing? Yes / No

hagre, 2025-03-04:
Just an idea for the settings. I know it looks like a lot (to much) of work, but maybe it can be used in some specific cases.

Make some settings possible to “inherit” from a base vehicle like prusa-slicer does it with the settings for 3d printers. 
Create a base vehicle with common antenna position, wheel base, … and inherit them to the actual vehicle with the possibility to modify just the changed settings. Mark the changes to show where it differs from the base vehicle (open lock). 

[image]

FantasyTeddy, 2025-03-04:
@richardklasens & @hagre:
I think these are quite valid suggestions and should probably be discussed as well.
However, my suggestion does not include anything else than the extraction of the vehicle-independent settings.
I would suggest to focus this issue on these system/environment settings and create new issues/discussions for your proposals.

richardklasens, 2025-07-29:
Lets fix this @FantasyTeddy

FantasyTeddy, 2025-08-19:
Just another thought regarding the settings:
If we are going to introduce a breaking change (which this most certainly will be), what about changing the storage format?

Historically, the format is XML with the format that *.settings files produce:
```xml
<configuration>
  <userSettings>
    <AgOpenGPS.Properties.Settings>
      <setting name="setWindow_Location" serializeAs="String">
        <value>989, 270</value>
      </setting>
    </AgOpenGPS.Properties.Settings>
  </userSettings>
</configuration>
```

What if we used XML, but in a format that is easier to serialize/deserialize:
```xml
<Settings>
  <WindowLocation>
    <X>989</X>
    <Y>270</Y>
  </WindowLocation>
</Settings>
```

Or maybe even change to something like JSON:
```json
{
  "Settings": {
    "WindowLocation": {
      "X": "989",
      "Y": "270"
    }
  }
}
```

Ultimately, we could also think about using a full-blown database as storage, which would have its own advantages/disadvantages.

What do you think?

richardklasens, 2025-08-19:
I think a database is cool but I'm not sure if necessary. The xml way you mentioned seems nice, json looks even better. And very easy and readable by human as well.

To not break it fully,
We can build a legacy importer which imports the old file, split it and ask user the new file names for env, vehicle and tool.

closed 2026-10-30 by richardklasens