AgOpenGPS-Official/AgOpenGPS #683: Proper ISOXML v3 generation

FantasyTeddy, 2025-03-04:
When validating the generated ISOXML v3 (`AgOpenGPS\Fields\<FieldName>\zISOXML\v3\TASKDATA.xml`) against the official XML schema ([ISO11783_TaskFile_V3-3.xsd](https://www.isobus.net/isobus/attachments/files/ISO11783_TaskFile_V3-3.xsd)) I get some validation errors.

Does someone have access to third-party software that reads ISOXML v3 files? Can you confirm (or deny) that the software reads the ISOXML from AgOpenGPS properly?

lansalot, 2025-03-04:
Try https://app.geo-bird.com/ for ISOXML file conversions, some useful tools in there.

FantasyTeddy, 2025-03-04:
Thank you for the link @lansalot. Unfortunately, I get an error when I try to import a field generated by AgOpenGPS into Geo-Bird. Can you do it successfully?