AgOpenGPS-Official/AgOpenGPS #708: Speed output in PGN is wrong

gunicsba, 2025-03-17:
I've noticed this with the speedpulse as well that we often have to set 13500 for a 100m.

Now with ISOBUS I've seen it on 2 different implement that there's a 5% or so error. 

Could we double check how that value is put on ethernet?

[image]

gunicsba, 2025-03-19:
Found the error, we manage to /27 instead of /27.7777 which results in the problem.

I asked Daan to fix it before we think about merging anything.
https://github.com/GwnDaan/AgOpenGPS/commit/938fcc197b0ec6446a89c77dc9506df7bf6c763c#r154000701

FantasyTeddy, 2025-03-20:
Great, I'm going to close this issue.

closed 2025-03-20 by FantasyTeddy