AgOpenGPS-Official/AgOpenGPS #722: Make boundary section detection lazier

gunicsba, 2025-03-30:
Issue: Our current settings is that as soon as the outer section touches the boundary it turns section off.
Which is what we expect but in reality this means driving on the boundary curve  with a real machine it turns the section on and off.

Not sure what would be the right solution but in reality I don't think anyone has under 25 cm sections so maybe that 10-20 centimeter of overshoot on the boundary would be acceptable?

For now we just force the sections on when driving the boundary curve.

I only moved the boundary curve 2 centimeter and the outer section is off:
[image]

Coverage was set to 50%

richardklasens, 2025-03-30:
If I’m right, Turn on the SectionFullyOutsideBnd button to turn off when the section is fully outside. It’s in the section width config next to the coverage setting.

gunicsba, 2025-03-30:
Yes Richard you're right but we do want it to be off when it's really outside. but when it's just barely outside it should stay on.

exentler, 2025-03-30:
I had similar issue when I was spreding fertiliser with my Bogballe M35W.
I used manual on the first outer round

P-R9, 2025-04-10:
Hello, 

why not make it as the most commertial Section Control Terminales? I also saw this issue by using Section Control and the border. So maybe the option I show here could be a solution.
In my opinion, the key word would be overlap tolerances in the settings of Sectioncontrol, similar to the example on the Amatron 4 below. 
 I also faced this problem because if driving with a Sprayer at the Headland Tramline you get the Problem that the last section deactivates if you have Boundary overlap at off and drive with 100% overlap in the case of herbicides. So in my opinion driver shouldn't press any workswitch at headland turning for that I have section control but as it is you don't drive perfect in the headland tramline and if you drive a bit far out or in curves which is normal in the field like gunicsbas exemple shows it would be nice if the system would think for the driver in this case. 

Best option would be a tolerance setting for the last sections in % or cm.
And another good change would be if you also could set the border overlap in % so it could be more individual, if you want it to be on set it to 100% and if off set it to 0% pretty easy to understand for everybody and it would be the same as for the normal overlap.

Exemples Amazone Amatron 4 overlap settings:

https://www.manualslib.com/manual/1618417/Amazone-Amatron-4.html?page=79#manual
https://www.manualslib.com/manual/1618417/Amazone-Amatron-4.html?page=80#manual

gunicsba, 2025-04-18:
[image]

mattmccormack, 2025-04-22:
@gunicsba that looks good. Using percent instead of absolute length avoids edge cases, like when the section width is smaller than the setting. Do you think it should be a separate setting?

P-R9, 2025-04-23:
@gunicsba I guess it isnt really importend if CM or % for me CM makes more sense in this setting but in't that importent more importend is that you can set is to small settings because in the most cases 30-50cm will be enough. 
Do you also want to change the boundary overlap to a % setting?
Because we often run into the contradiction that with 3 Meter Sections 0% makes big blind spots but if we do 100% you get problem with de goverment so to set it up in between would be great. I think these changes would be a big benefit for Section Control if you want it to use in the headland and boundary too, which should be the goal if we have RTK in my opinion. But maybe that is just advantage in regions where you don't have big fields and as a driver you are doing mostly headland with 27 meter Sprayer :D

richardklasens, 2025-07-29:
Closed as no response on the PR