AgOpenGPS-Official/AgOpenGPS #758: Tool Lift look ahead timer problem

Pat-I, 2025-04-20:
Some users say v6.6 and 6.7 don't trigger the tool lift at the right moment.
It seems to trigger only when the tool passes, not when the pink timer line crosses.
v6.4.4 seems fine.

link to the discution on Discourse:
https://discourse.agopengps.com/t/hyd-lift-look-ahead/19191?u=pat

mattmccormack, 2025-04-20:
I can replicate the problem but I don't understand why they said 6.4.4 worked, because I think the bug has existed for longer than that. I wonder if the difference was whether the headland was set to section controlled or not? It works correctly when **not** set to section controlled.

[image]

If headland **is** set to section controlled, the implement is lowered just as the tool crosses the headland, ignoring the look ahead value.
[image] 

With either section control options, the hydraulics are raised only when the tool reaches the headland. This part is correct, isn't it?

mattmccormack, 2025-04-21:
For anyone else who was confused like I was:

| Icon | File Name | Resource Name | `bnd.isSectionControlledByHeadland` |
| --- | --- | --- | --- |
| [image] | HeadlandSection**Off**.png | HeadlandSection**On** | true |
| [image] | HeadlandSection**On**.png | HeadlandSection**Off** | false |

If you had it set to paint the headland, wouldn't you also want the implement to already be in the ground by then? So the implement should lower when the pink line crosses the outer boundary, not the headland line. Otherwise, why are you painting the headland if the implement is out of the ground?

I have a quick fix but I want to understand the requirements properly before submitting a PR since I might be way off.

```diff
diff --git a/SourceCode/GPS/Forms/OpenGL.Designer.cs b/SourceCode/GPS/Forms/OpenGL.Designer.cs
index 5075298..27679a2 100644
--- a/SourceCode/GPS/Forms/OpenGL.Designer.cs
+++ b/SourceCode/GPS/Forms/OpenGL.Designer.cs
@@ -816,7 +816,7 @@ private void oglBack_Paint(object sender, PaintEventArgs e)

                 //draw 250 green for the headland
-                if (bnd.isHeadlandOn && bnd.isSectionControlledByHeadland)
+                if (bnd.isHeadlandOn)
                 {
                     GL.LineWidth(3);
                     GL.Color3((byte)0, (byte)250, (byte)0);
@@ -898,7 +898,7 @@ private void oglBack_Paint(object sender, PaintEventArgs e)
             //10 % min is required for overlap, otherwise it never would be on.
             int pixLimit = (int)((double)(section[0].rpSectionWidth * rpOnHeight) / (double)(5.0));

-            if ((rpOnHeight < rpToolHeight && bnd.isHeadlandOn && bnd.isSectionControlledByHeadland)) rpHeight = rpToolHeight + 2;
+            if ((rpOnHeight < rpToolHeight && bnd.isHeadlandOn)) rpHeight = rpToolHeight + 2;
             else rpHeight = rpOnHeight + 2;

             if (rpHeight > 290) rpHeight = 290;
```

Pat-I, 2025-04-21:
> If you had it set to paint the headland, wouldn't you also want the implement to already be in the ground by then? So the implement should lower when the pink line crosses the outer boundary, not the headland line. Otherwise, why are you painting the headland if the implement is out of the ground?

I'm pretty sure Lift control should only be active when Headland is active. If no headland, you don't want automatic lift/lowering

As for painting, Automatic lift/lower must NOT be linked with painting in ANY way. Painting is controlled by the workswitch and the headlands and/or boundary lines.

RollingHills540, 2025-04-22:
Correct version was not the difference. It does as described, set to apply headland the timer does not affect. But if selected to not apply to headland it works as expected.

mattmccormack, 2025-04-22:
Thanks for checking. 

Is there ever a time where you would have "Apply to Headland" active at the same time as using automatic lift/lower? 

I think the current behaviour is unintuitive. Should we change it so the lookahead is always used for lowering, regardless of setting?

RollingHills540, 2025-04-23:
I don't know when it would be required to have apply to healdand active while using the lift function. 

I would say it is just figuring the flow out. I'm a fairly new user just getting the hang of it. I just assumed with the lift active and the timer line showing up it would use the timer, but knowing the headland application makes a difference I can do what I need.

Pat-I, 2025-04-27:
@mattmccormack @RollingHills540 
Can you confirm that is lookahead work as expected when not painting the headland?

This "painting Headland" button was added later. Probably the consequences weren't studied enough. What is the use for this? Lifting when you leave the field? That is the only thing I see it does.

Pat-I, 2025-04-27:
> I think the current behaviour is unintuitive. Should we change it so the lookahead is always used for lowering, regardless of setting?

I would think so, but it does only trigger at the boundary from what I see and I guess you don't want it to work to soon there.

Pat-I, 2025-04-27:
[image]
I think This button (setting) is useless and should be removed

richardklasens, 2025-04-27:
> [image] I think This button (setting) is useless and should be removed

It’s not useless. If you just want to use the headland as just a line without anything else.. it works fine.

RollingHills540, 2025-04-27:
I will run it again as soon as possible.

I see the function of the not apply to headland. Maybe in wet conditions I'd plant the main field before the headland then complete the headland after, just a thought. Makes sense if other functions were added later that something odd like this would creep in.

Pat-I, 2025-04-28:
> It’s not useless.

Yes, indeed. It was explained to me that this is for section control and not directly related to autolift.

So one question remain:
then it's set to apply to headland, does autolift should use a look-ahead when lowering? Knowing that the time it trigger is when you cross a boundary (outer and inner too I guess)

RollingHills540, 2025-04-28:
I think the look ahead timer should work regardless of the apply to headland option. Primarily because the tool wouldn't be lowered in time to work the main inner passes, so it would miss more when apply to headland is selected. I would assume it was just an error adding new features, maybe I'll understand the code more in the future to be of better help.

mattmccormack, 2025-04-28:
> I think the look ahead timer should work regardless of the apply to headland option.

I agree with this. If there is a look-ahead time set, then it should be used. I've done this in pull request #769. Would appreciate a review if anyone has time. 

---

I still can not think of a time when you would want these two settings set simultaneously, but if someone wants to use it like that they can:
[image]
(auto hyd lift ON and paint headland ON)

Pat-I, 2025-05-17:
Should be resolved by https://github.com/AgOpenGPS-Official/AgOpenGPS/pull/769#issue-3024609383

closed 2025-05-17 by Pat-I