AgOpenGPS-Official/AgOpenGPS #759: Setup code analysis

gunicsba, 2025-04-20:
Creating this ticket based on the discussion you had under @richardklasens  AgShare PR.

I agree that there should be formatting and coding standards that we (the broader community who adds 5-20 lines of code the whole year to the project) could easily follow.
Having this part of the project and builds would make things simpler on the longer run.

With that being said I'm sure most of the existing source code should be rewritten.

Would it be possible to set up code analysis in an optional / temporary way?
So new PR-s could be checked and eventually we'd fix all the existing code?

FantasyTeddy, 2025-04-20:
Thank you for creating this issue.

> Would it be possible to set up code analysis in an optional / temporary way?
So new PR-s could be checked and eventually we'd fix all the existing code?

Unfortunately, I don't think there is an easy way to do that.

However, I think it is finally time to bite the bullet and do all those major refactorings (like #570 and running code formatting) in one big PR, such that we can start benefiting from the value that would bring us.
I don't think that it is worth waiting too much longer. All the changes already happening (like the big MVVM refactor from @J005t67) have a much bigger impact on the overall code structure, than moving some files around and/or formatting code.

FantasyTeddy, 2025-04-22:
@AgOpenGPS-Official/aog-coders:

Can you give a quick feedback if we should do any (or both) of these options now:

- **A**) Organize the folder structure of the project (see #570)
- **B**) Introduce code analysis and run code formatting over the entire codebase

FantasyTeddy, 2025-05-07:
> +1 to the code analysis, but the folder structure, I'd say leave that for the time being - or at least set a date for people to have work committed and merged, so it causes less trouble when the time comes?

Could we do it the other way around? Instead of me setting an arbitrary date, you tell me when would be a good opportunity to do it?
Who has pending work that still needs to be merged? I know of the PRs from @GwnDaan regarding the ISOBUS integration (#607) and the AgShare implementation from @richardklasens (#748). Anything else?

closed 2026-10-30 by richardklasens