AgOpenGPS-Official/AgOpenGPS #790: Headland autosteer disengage

heinzimus, 2025-07-08:
When you go past the headland, disengage the autosteer, for some machines that require complex headland maneuver don't have to press the button every time.
Has to be after or while the implement rise, without cancelling

heinzimus, 2025-07-08:
Steering motor attach to the steering wheel via timing belt, its a 12volt motor. But i have to put the current limit at max or it trip when turning fast. And when I override manually turning it's a bit heavy and like 1 sec until disengage, so I prefer to unlock with the button. That's why I am asking for this feature. Thanks

richardklasens, 2025-07-27:
Hi,

Sorry to let you know but this won't be added at the moment.

closed 2025-07-27 by richardklasens